liamcottle/reticulum-meshchat #2: WEB.PY in windows CLI error

mrmate2, 2024-05-11:
File "C:\Users\ANATOLIY\reticulum-webchat\web.py", line 249, in run
    asyncio.get_event_loop().add_signal_handler(signal.SIGINT, lambda: exit(-1))
  File "C:\Users\ANATOLIY\AppData\Local\Programs\Python\Python312\Lib\asyncio\events.py", line 582, in add_signal_handler
    raise NotImplementedError
NotImplementedError

When I try to run web.py on win11, Iget above error.
how to fix?

liamcottle, 2024-05-16:
I didn't run into this error on my Windows 10 machine, but when I ran it on Windows 11 I did indeed see this issue.

I've just pushed a commit to remove the signal handler, since this isn't supported on Windows 11, and the shutdown handler I setup seems to work as expected anyway.

The signal handler code was added during debugging, so probably doesn't need to stick around anyway.

Please let me know if you're able to get it working on your end, and we can close this :)

mrmate2, 2024-05-16:
Dear Sir, this is to confirm that in Windows11 WEB.PY script now starts without problem and it gives acces to server web page.
This issue can be closed.
The only next problem for me that a little unclear what to do next on that web page? It Writes "connected", but on the display of rNode there is no activity, tge display is the defoult one, whicр normally we have just aftre flashinп firmware to rNode.
[image]
[image]
[image]

liamcottle, 2024-05-16:
> this is to confirm that in Windows11 WEB.PY script now starts without problem

That's fantastic! Glad it's working for you now.

> what to do next on that web page

If you press the announce button in the top right, an announce will be sent to all enabled interfaces you configured in your reticulum config file.

After that, you just have to wait until announces are received from other users on the network. Once received, you can click on a peer and start chatting.

At this stage, there is no web ui for managing the interfaces used in Reticulum, for now you still have to edit the config file like you normally would.

Make sure you have added your RNode into the config file, and when you press the announce button in the web ui, you should see the LED on the RNode light up. This indicates that it transmitted the announce over the radio.

You could try adding one of the test net servers to your config, and within a few minutes you'll likely see other peers/nodes show up.

Here's an example config from one of my test setups.

```
[reticulum]
  enable_transport = True
  share_instance = Yes

[logging]
  loglevel = 1

[interfaces]
  
  ## Default interface does not work on Windows
  [[Default Interface]]
    type = AutoInterface
    enabled = True

  # RNode interface for LoRa in New Zealand
  [[RNode LoRa Interface]]
    type = RNodeInterface
    interface_enabled = True
    port = /dev/cu.usbserial-0001
    # Set frequency to 917.375 MHz (slot 10 Australia/New Zealand)
    frequency = 917375000
    # Set LoRa bandwidth to 125 KHz
    bandwidth = 125000
    # Set TX power to 22 dBm (~150mW)
    txpower = 22
    spreadingfactor = 8
    codingrate = 5

  # TCP/IP interface to the RNS Amsterdam Hub
  [[RNS Testnet Amsterdam]]
    type = TCPClientInterface
    enabled = True
    target_host = amsterdam.connect.reticulum.network
    target_port = 4965
```

mrmate2, 2024-05-16:
Oh, I see, need to edit config file.
Many thanks for detailed instructions!

closed 2024-05-16 by mrmate2

liamcottle, 2024-05-16:
No problem! Happy to help.

Don't forget to stop and restart the `web.py` script when you edit your config file. I don't think Reticulum picks up changes automatically.

Also, you may want to check out the readme, for [using a persistent identity file](https://github.com/liamcottle/reticulum-webchat?tab=readme-ov-file#using-an-existing-reticulum-identity), as currently, you will get a new identity every time you run the script.