liamcottle/reticulum-meshchat #4: Windows Portable doesn't accept same parameters as web.py

vitorio, 2024-06-14:
For the portable Windows executable to really be portable (e.g. run off an encrypted volume on a USB stick), it'd be useful for it to accept the same parameters as the web.py launcher so all the identity, configuration, and storage can be stored alongside it.  Right now, it's hardcoded (in [main.js](https://github.com/liamcottle/reticulum-meshchat/blob/master/electron/main.js)) to `~/.reticulum-meshchat` for `storageDir` with no option to set `configDir`.  It'd be useful to accept these:

```
  --identity-file IDENTITY_FILE
                        Path to a Reticulum Identity file to use as your LXMF address.
  --identity-base64 IDENTITY_BASE64
                        A base64 encoded Reticulum Identity to use as your LXMF address.
  --reticulum-config-dir RETICULUM_CONFIG_DIR
                        Path to a Reticulum config directory for the RNS stack to use (e.g: ~/.reticulum)
  --storage-dir STORAGE_DIR
                        Path to a directory for storing databases and config files (default: ./storage)
```

In the interim, a workaround is to extract the Electron app, edit `resources/app/electron/main.js`, and change [line 83](https://github.com/liamcottle/reticulum-meshchat/blob/master/electron/main.js#L83) to modify `storageDir` to the desired path.  Then, add code like this below it to set `configDir` to the desired path:

```javascript
        // determine path for config
        const configDir = path.join('.reticulum'); // .reticulum
```

Finally, change the `exeChildProcess` call to:

```javascript
        // spawn executable
        exeChildProcess = await spawn(exe, [
            '--headless', // reticulum meshchat usually launches default web browser, we don't want this when using electron
            '--port', '9337', // FIXME: let system pick a random unused port?
            '--storage-dir', storageDir,
            '--reticulum-config-dir', configDir,
        ]);
```

to use the new `configDir` path.  Then, just run the extracted `Reticulum MeshChat.exe` directly instead of the packaged portable file, and it'll use the modified JS.

liamcottle, 2024-07-09:
Hey! Thanks for reporting this, I haven't forgotten about it. Just been a bit busy!

I want to add a UI for managing config paths etc, but for now it's probably easier to just passthrough any command line args passed to the `.exe` and merge them in to the default provided in `exeChildProcess`.

Will plan to sort this before the next release.

liamcottle, 2024-07-09:
This is now supported and has been merged into the main branch with commit a804e16f3d28d011d410ca017eacbc7a97b6187f

I tested this on my Windows machine by creating a shortcut link to the portable `.exe` with a custom `--storage-dir` and `--reticulum-config-dir`. I did have to provide an absolute path and not a relative path, otherwise the folders were created elsewhere. Unsure where, because the shortcut had the working dir set to where it was opened from...

I have not tested this on an external USB drive yet, but will attempt this later on. Let me know if this works for you and I can close this issue :)

closed 2024-07-31 by liamcottle