liamcottle/reticulum-meshchat #10: Show some feedback after "announce"

trosel, 2024-08-16:
There seems to be a lot of time in between pressing announce and actually seeing anything in the UI (if at all, I'm still waiting).

Could there be some kind of feedback in the UI that something is happening behind the scenes after you click "announce"? I don't have any ideas to offer here. But It does seem like it doesn't work when you hit announce and nothing happens.

liamcottle, 2024-08-16:
Hey there! Thanks for trying out MeshChat :)

When you click the announce button in the bottom left corner, there is a label underneath that says when the last announce happened.

Is this not updating for you when you click announce?

trosel, 2024-08-17:
It does update, but nothing in the UI changes. My understanding is that eventually other users should start to show up in the UI. So maybe some kind of feedback like "searching for users, none found yet" or even a "searching for users, this could take a minute, or forever" idk haha.

Or maybe this is just a request for an onboarding flow that explains what to expect and how to ensure that the app is usable (there's people to talk to)?

liamcottle, 2024-08-19:
Fair enough!

> Or maybe this is just a request for an onboarding flow that explains what to expect and how to ensure that the app is usable (there's people to talk to)?

I do plan to build an onboarding/introduction screen to explain how things work, but the app is under heavy development so I don't want to do this just yet, as screenshots will be outdated pretty quick and I'm still thinking about what to write.

> So maybe some kind of feedback like "searching for users, none found yet" or even a "searching for users, this could take a minute, or forever" idk haha.

If you look at the announces tab on the messages page, you'll see that it mentions it's waiting for someone to announce. If no one announces, you'll never see anything here.

The way MeshChat is intended to be used, is to create your own mesh network to communicate over.

By default, the only interface that is configured is the `AutoInterface`. This allows you to communicate with anyone else that is on the same LAN, (that also has an `AutoInterface`). They could be using MeshChat, or something else like Sideband or Nomadnet.

If you want to talk to other people in the world, you need to establish your own mesh network. This is done by adding interfaces. For testing purposes, you could add the Reticulum Test Net, which is a `TCPClientInterface`. This would allow you to then communicate with anyone else that also has that interface configured. Basically, you are connecting to a shared TCP server, which forwards your traffic to other peers.

You can also optionally enable your MeshChat instance to be a transport node. This will allow anyone else connected to your interfaces to communicate with anyone else on the rest of your interfaces. Being a transport node, basically means you transport packets for other peers between all of your interfaces.

There is no UI toggle for this yet, but one will be added in the future. For now, you have to edit the reticulum config file.

trosel, 2024-08-20:
> For testing purposes, you could add the Reticulum Test Net, which is a TCPClientInterface.

In an effort to bootstrap usage, maybe this could be made easier. Basically, let people use LXMF like a normal internet chat app and then fall back to mesh networks. Rather than how (I think) it is today where you use LXMF as a mesh network chat app and then fall back to normal internet.

Basically have some sane default that allows people to actually use the app right away and then let them configure more advanced options as needed later on. Let them discover what "mesh" means organically over time rather than requiring them to learn everything up front before the app is usable.