liamcottle/reticulum-meshchat #12: audio codec

ramon159, 2024-08-25:
is possible switch between two codec protocols?
and add support for more one like MELPe?

look the difference:
https://m-k.mx/static/codec2/

markqvist, 2024-09-24:
LXMF itself specifies audio modes for Codec2 and OPUS audio currently.

- MeshChat supports Codec2 at 1200 and 2400 bps
- Sideband supports Codec2 at 1200, 2400 and 3200 bps, as well as OPUS in all modes, but currently only encodes in the VoIP profile.

MELPe does sound very nice, but since the MELPe codec itself is proprietary and subject to a slew of commercial licensing issues, I don't personally see it being added as a supported codec in LXMF.

liamcottle, 2024-11-01:
Hey! I don't have any experience with MELPe, but as @markqvist mentioned it's proprietary and probably wouldn't be added as a supported LXMF codec. Due to that, it probably wouldn't make it's way into MeshChat either.

But to answer your question on supporting multiple codecs, the most recent version of MeshChat now supports high quality opus voice messages (not audio calls yet). So at this stage, Codec2 and Opus are supported in MeshChat.

Closing this out as not planned, for MELPe :)

closed 2024-11-01 by liamcottle