liamcottle/reticulum-meshchat #13: Android 14 access localhost:0000

cyclingtogether, 2024-09-18:
Successfully built in android/termux.

Startet meshchat without errors but not able to access localhost:8000 from any webbrowser.
Probably a security feature in android 14?

How would you access it, any hints?

note:
(installing and starting firefox out of termux is missing an x environment, so this does not work)

liamcottle, 2024-09-24:
Hey! Unsure on this one sorry. I'm also on Android 14, but I was able to build and run on Termux without any issues. I could access with https://locahost:8000 via Google Chrome and also an installed PWA.

No errors in the console in termux?

cyclingtogether, 2024-09-26:
no compile errors. i have a highly [debloated](https://github.com/0x192/universal-android-debloater) phone, maybe its missing some invasive google thingie.. will tell when figured out or someone points me to the missing piece

magdesign, 2024-10-26:
maybe this works only on a rooted device?

having a samsung with oneui, there seems to be a hidden firewall. would nice to know of anyone else can access localhost from any browser.

liamcottle, 2024-12-29:
I'm going to close this out as I haven't seen any follow ups, and it's working on my Samsung S24 with Android 14, but if this is still an issue, try the following:

```
python3 meshchat.py --headless --host 0.0.0.0
```

closed 2024-12-29 by liamcottle

magdesign, 2024-12-29:
For now i run it on my linuxmobile phone since it is still not possible to view localhost on a samsung device... Hopefully we will have more people wanting to use this on a proprietary Android device... to find a solution.

thank you for making mashchat!

magdesign, 2025-02-27:
user l78193 on matrix asked what errors i get when running:

`nc -lv 5555`

in termux and then opening a browser `127.0.0.1:5555`

in termux i get:

```
Listening on 0.0.0.0 5555
Connection received on localhost 55762
GET / HTTP/1.1
Host: 127.0.0.1:5555
Connection: keep-alive
sec-ch-ua: "Not(A:Brand";v="99", "Google Chrome";v="133", "Chromium";v="133"
sec-ch-ua-mobile: ?1
sec-ch-ua-platform: "Android"
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Linux; Android 10; K) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/133.0.0.0 Mobile Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
Sec-Fetch-Site: none
Sec-Fetch-Mode: navigate
Sec-Fetch-User: ?1
Sec-Fetch-Dest: document
Accept-Encoding: gzip, deflate, br, zstd
Accept-Language: en-IE,en;q=0.9,de-CH;q=0.8,de;q=0.7,en-GB;q=0.6,en-US;q=0.5
```

magdesign, 2025-02-27:
😂😂😂😂

finnaly solved this with:

`npm run build-frontend`

maybe spitting a warning in the terminal would be nice when frontend was not build

adingbatponder, 2025-02-28:
I too cannot connect to <ipaddress>:8000 via Chrome/Firefox on Sony Xperia running Android 15  .... a real shame indeed. The same <ipaddress>:8000 works fine on a computer (firefox on Ubuntu 22)