liamcottle/reticulum-meshchat #16: Sending Failed: 'LXMRouter'

magdesign, 2024-10-03:
Still trying to figure out why i can receive messages, but only seldom able to send some. Yesterday i thought its a firewall problem, but now i get error messages:
[image]

liamcottle, 2024-10-03:
Hey @magdesign, looks like you've installed from source, and then later updated to the most recent commits.

You'll need to pull in the updated python dependencies, which includes updates to RNS and LXMF. There are a few new features implemented which require the new RNS version, and it's complaining that the attribute doesn't exist (since using older version).

```
pip install -r requirements.txt
npm install --omit=dev
npm run build-frontend
```

The above should get you updated correctly. I'll need to add a couple of helper scripts for easily installing and updating via Termux :)

magdesign, 2024-10-03:
Thanks! 🙏
removing RNS and LXMF and then installing the newest versions with

```
pip install rns --break-system-packages
pip install lxmf --break-system-packages
```
and then building as you mentioned helped to get rid of the error.

closed 2024-10-03 by magdesign