liamcottle/reticulum-meshchat #19: Radio State Mismatch RAK4361

ToxicityKingETN, 2024-11-07:
Radio state mismatch after verified identical with RAK4361
Radio Settings
[image]
Meshchat Settings
[image]
Meshchat response
[image]

liamcottle, 2024-11-08:
Hey there!

I don't see any changes between v1.77 and v1.78 of RNode firmware that could cause this:
https://github.com/markqvist/RNode_Firmware/compare/1.77...1.78

But based on your screenshots, it looks like you are running MeshChat from source.
It could be possible that you are running older commits, or you have not updated the Python dependencies.

The only thing I know of that causes the mismatch, is that some older RNS versions would limit the TX power to 17, even though the radio was configured for and supported 22.

If you could try updating the MeshChat source, and deps with these command that would be great:

```
git pull
pip install -r requirements.txt
```

ToxicityKingETN, 2024-11-08:
I did as you said and tried lowering the power to 17, both repeat the same issue, unfortunately

ToxicityKingETN, 2024-11-26:
Here is another attempt, verifying interface config,
rnode config:
[image]
rns interface config(interface is true):
[image]
Meshchat error:
[image]

markqvist, 2024-12-06:
If you are controlling the RNode via `rnsd`, or any other application that uses Reticulum, you should not put the RNode into TNC mode (as your logs suggest you have done). TNC mode is only for using the RNode with applications that don't support setting the frequency and modulation parameters themselves!

Also, this *might* be related to a bug in the RAK builds of the `1.78` version of the firmware, that @jacobeva discovered and is working on (so just pinging you here Jacob, for any info this may provide you).

If this is indeed that bug, hopefully we'll have an update out very soon that fixes it.

jacobeva, 2024-12-06:
https://github.com/markqvist/RNode_Firmware/issues/95, please close this issue.

closed 2024-12-06 by ToxicityKingETN