liamcottle/reticulum-meshchat #30: Flatpak Package

LinuxinaBit, 2024-12-25:
It would be really cool to have a Flatpak package of MeshChat!

Generally the OS shouldn't matter (as long as it's Linux), the entire point of Flatpak is there should never be a "but it works on my machine" situation.
They list many of the supported distros here: https://flatpak.org/setup/ and I am currently using Fedora Budgie Atomic (https://fedoraproject.org/atomic-desktops/budgie/).

Flatpak Getting Started page: https://docs.flatpak.org/en/latest/getting-started.html
General Flatpak docs: https://docs.flatpak.org/en/latest/

LinuxinaBit, 2024-12-25:
Oh, also maybe publish it on Flathub:
https://flathub.org/

Docs: https://docs.flathub.org/docs/for-app-authors/submission/

fotografm, 2025-01-08:
I am pleased to see that it is already in  the Arch User Repository AUR :-)