liamcottle/reticulum-meshchat #34: FATAL:setuid_sandbox_host.cc(158)

ronnietucker, 2024-12-29:
Running the AppImage on TuxedoOS (aka: Ubuntu with KDE) and I get:

`[31262:1229/191655.761245:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_ReticuJQ7lJt/chrome-sandbox is owned by root and has mode 4755.`

liamcottle, 2024-12-30:
Howdy! There's some previous discussion around this in the Matrix chat.
https://matrix.to/#/!TRaVWNnQhAbvuiSnEK:matrix.org/$JwLq3WF5jGgzEF9Du6e85jrnEtnCFiq3hBECzP9CrN8?via=matrix.org&via=tchncs.de&via=envs.net

I also saw this issue when trying to run the AppImage from CLI, but double clicking it from GUI seemed to launch fine.

If you find a way to resolve this, it would be great to get it fixed properly :)

liamcottle, 2024-12-30:
https://github.com/user-attachments/assets/22170512-8855-4799-9581-7d03f73cb3c4

ronnietucker, 2024-12-30:
Thanks for the info. Here's what I've got so far:

If I double click the AppImage (in KDE) it won't run at all. Looks like it's doing nothing.

If I try and run it from the terminal I get the error message in my original post.

If I use AppImageLauncher and double click it it'll ask if I want to run it once, or integrate it.
If I run it once, it does nothing again.
If I click to integrate, then choose it from my KDE menu it will run, but then I get:
[image]

liamcottle, 2024-12-30:
Thanks for the info, could you have a read through this one, and check the kernel settings?

https://docs.appimage.org/user-guide/troubleshooting/electron-sandboxing.html

If this doesn't help, I guess the only option is to add that flag to disable the sandboxing... In your screenshot it looks like it gets passed to my python script instead of the electron process.

Let me know if the kernel settings make a difference, if not, I'll look at adding the flag internally.

ronnietucker, 2024-12-30:
Yep, my kernel replies with a 1 (as I think it's meant to?)

If I use the terminal to run Reticulum (the gibberish before AppImage is just part of AppImageLauncher) with the -no-sandbox it does run, but like last time it crashes out and I see:

`$  ./ReticulumMeshChat-v1.17.0-linux_e1784ecf710d3e9b81182c4c46e6a50b.AppImage -no-sandbox`
`usage: ReticulumMeshChat [-h] [--host [HOST]] [--port [PORT]] [--headless]`
`                         [--identity-file IDENTITY_FILE]`
`                         [--identity-base64 IDENTITY_BASE64]`
`                         [--generate-identity-file GENERATE_IDENTITY_FILE]`
`                         [--generate-identity-base64]`
`                         [--reticulum-config-dir RETICULUM_CONFIG_DIR]`
`                         [--storage-dir STORAGE_DIR]`
`                         [--test-exception-message TEST_EXCEPTION_MESSAGE]`
`                         ...`
`ReticulumMeshChat: error: unrecognized arguments: -no-sandbox`

`[13017:1230/114730.271686:ERROR:browser_main_loop.cc(280)] GLib-GObject: ../../../gobject/gsignal.c:2685: instance '0x345000ef6110' has no handler with id '5168'`

liamcottle, 2024-12-30:
Cool, thanks for checking that. I will need to spin up a VM to investigate this further. This will likely take me a couple of days. I'll update you once I have a fix :)

ronnietucker, 2024-12-30:
Just to add that while the error says:
`You need to make sure that /tmp/.mount_ReticuAWHeOM/chrome-sandbox is owned by root and has mode 4755.`

I don't see any .mount_Reticu... folder in /tmp/

stephen304, 2025-04-18:
@liamcottle I was looking at other projects that have the same issue, and [according to another dev](https://github.com/zoidsh/meru/issues/419#issuecomment-2775318588), this issue was fixed in electron-builder v26, so it seems like [simply bumping that version](https://github.com/zoidsh/meru/commit/ba3f1da08089a9a62c03fb5c502b35712be9e8b9) might be all that's needed.

Nao-30, 2025-05-09:
> Running the AppImage on TuxedoOS (aka: Ubuntu with KDE) and I get:
>
> `[31262:1229/191655.761245:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured [...]

Ubuntu 23.XX and higher versions  implemented a stricter security model through AppArmor that specifically targets unprivileged user namespaces[2](https://blog.mohammed-al-kebsi.space/post/understanding-ubuntu-24-04-appimage-sandbox-restrictions/).

you can review the full article

For a complete understanding of this issue, including technical details, implementation specifics, you can take a look at:

[Technical Deep Dive](https://blog.mohammed-al-kebsi.space/post/understanding-ubuntu-24-04-appimage-sandbox-restrictions/)
For a more direct practical solutions, you can take a look at:

[Practical Workarounds](https://blog.mohammed-al-kebsi.space/post/fix-appimage-sandbox-issues-ubuntu-24-04/)