liamcottle/reticulum-meshchat #37: EndeavourOS Rnode USB access denied

fotografm, 2025-01-03:
I got reticulum-meshchat working fine on Linux Mint 22 with Heltec v3 Rnode. My problem is trying to get it to work on EndeavourOS. My reticulum-meshchat RnodeInterface reports "disconnected" no matter what I do. On Mint I had to add my user to the dialout group and it worked. EndeavourOS can communicate successfully with the node over USB as I used sudo rnodeconf /dev/ttyUSB0 --autoinstall and it succeeded with no errors. Any suggetion as to how to overcome the problem ?

liamcottle, 2025-01-03:
Hey there! I got your email with the extra info, but I've decided to reply here so this information is public in case someone else runs into the same issue.

I don't have a bare metal machine that I can put EndevourOS onto right now, so instead, I downloaded the ISO and launched a live instance as a VM on my MacBook via Parrallels Desktop.

I was able to get the RNode to work in MeshChat in EndevourOS without needing to make any changes to the OS, just following the steps below it worked fine first go:

- Launched Firefox
- Downloaded MeshChat.AppImage from this repo GitHub
- Double clicked to launch it.
- Went to Interfaces page and added an RNodeInterface and selected the `ttyACM0` serial port.
- I configured valid settings for the RNode, saved it, and then restarted MeshChat.
- Once relaunched, the interface was connected, and I was able to announce via LoRa.
- I received this announce on another machine with another LoRa device, so have confirmed it works.

I then downloaded Google Chrome via a terminal by running `yay google-chrome`. After loading the the web flasher site and clicking the `Detect RNode` button, I did indeed have issues getting access to the serial device in the browser. I was receiving `Network device has been lost` error messages in the dev tools console.

I looked into adding my user to `dialout`, but that group didn't exist. After some searching, I found this StackOverflow answer, which resolved the issue for me. https://stackoverflow.com/a/47691019

Run `lsusb` and grab the device info from the output. Then you'll need to create a udev rule. I've attached screenshots of this below. Once you've added the rule, close Google Chrome, and unplug and replug your device. It worked for me after relaunching the web flasher.

Please let me know if this fixes the issue for you :)

https://github.com/user-attachments/assets/1ea3a641-69bd-4767-9199-aefa5cd9c7f8

[image]

[image]

[image]

fotografm, 2025-01-03:
Hi Liam, Thanks for your quick and detailed reply. I followed what you did but when I add interface to Meshchat, I still get the result for RNode as "disconnected" in red colour. lsusb shows the Heltec Rnode as /dev/ttyUSB0. ACM0 is my Fibocom WWAN adapter I think.
I went on to create the udev rule as you showed and rebooted just in case but Chrome RNode flasher - detect RNode fails, with the developer console error "Failed to open serial port".
I get on chrome://device-log  (unrelated entries not shown here):
Serial Error [14:14:52] Failed to open serial port: FILE_ERROR_ACCESS_DENIED
Serial Event [14:14:48] Serial device added: path=/dev/ttyACM0 vid=2cb7 pid=0002 serial=004999010640000
Serial Event [14:14:48] Serial device added: path=/dev/ttyUSB0 vid=10c4 pid=ea60 serial=0001
USB User [14:13:55] USB device added: path=/dev/bus/usb/001/002 vendor=4292 "Silicon Labs", product=60000 "CP2102 USB to UART Bridge Controller", serial="0001", guid=86ddb249-d22c-4a49-b2b8-dd14585c12d6
I plugged the Rnode into my Mint22 machine and everything works fine there as expected.
Obviously there must be some diffreence between your Endeavour VM and my bare metal install on a Thinkpad. What do you suggest I could try next ?

fotografm, 2025-01-03:
Also, If I start the Meshconsole at the command line with ./ReticulumMeshChat...AppImage then I repeatedly get the command line error report "Permission denied /dev/ttyUSB0". I also tried
sudo udevadm control --reload-rules
sudo edevadm trigger
to no avail.

fotografm, 2025-01-04:
I have got it to work now on EndeavourOS. The solution is uncomplicated:
sudo chmod a+rw /dev/ttyUSB0
I can now access the RNode via USB. Probably this fix won't survive a reboot so I will try a udev rule to fix this.

liamcottle, 2025-01-05:
Glad you got it working :)

closed 2025-01-05 by liamcottle

fotografm, 2025-01-05:
Thanks for your help. I wanted to add my solution for surviving a reboot
but the case is closed.
I just created a udev rule that ensures access to USB0 after a reboot of
EndeavourOS

sudo nano /etc/udev/rules.d/002.rules
SUBSYSTEM=="usb", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60",
MODE="0666", GROUP="wheel"
KERNEL=="ttyUSB*" , GROUP="wheel" , MODE="0666"