liamcottle/reticulum-meshchat #38: Is there a command payload sending option?

Terramoto, 2025-01-03:
Hello,

It looks like in lxmf payloads there's a section fields that is usually used to be interpreted as a command rather than just test but the option appears to be missing on meshchat.

Command payload:
```
[2025-01-03 21:34:11] [Debug] LXMF - Message received:
[2025-01-03 21:34:11] [Debug] -   Date/Time: 2025-01-03 21:34:13
[2025-01-03 21:34:11] [Debug] -       Title:
[2025-01-03 21:34:11] [Debug] -     Content:
[2025-01-03 21:34:11] [Debug] -      Fields: {9: [{0: 'ola'}]}
[2025-01-03 21:34:11] [Debug] -        Size: 31 bytes
[2025-01-03 21:34:11] [Debug] -      Source: <765ca1763e4eb5453234f659cf55a782>
[2025-01-03 21:34:11] [Debug] - Destination: <c715110b3da697e7d665f2bb90bdf134>
[2025-01-03 21:34:11] [Debug] -   Signature: Validated
[2025-01-03 21:34:11] [Debug] -    Attempts: 0
[2025-01-03 21:34:11] [Debug] -      Method: 3
[2025-01-03 21:34:11] [Debug] LXMF - Call to registered message received callback
[2025-01-03 21:34:11] [Notice] cmd: ola
```
Message payload:
```
[2025-01-03 21:34:19] [Debug] LXMF - Message received:
[2025-01-03 21:34:19] [Debug] -   Date/Time: 2025-01-03 21:34:21
[2025-01-03 21:34:19] [Debug] -       Title:
[2025-01-03 21:34:19] [Debug] -     Content: ola
[2025-01-03 21:34:19] [Debug] -      Fields: {}
[2025-01-03 21:34:19] [Debug] -        Size: 8 bytes
[2025-01-03 21:34:19] [Debug] -      Source: <765ca1763e4eb5453234f659cf55a782>
[2025-01-03 21:34:19] [Debug] - Destination: <c715110b3da697e7d665f2bb90bdf134>
[2025-01-03 21:34:19] [Debug] -   Signature: Validated
[2025-01-03 21:34:19] [Debug] -    Attempts: 0
[2025-01-03 21:34:19] [Debug] -      Method: 3
[2025-01-03 21:34:19] [Debug] LXMF - Call to registered message received callback
```

It would appear the command is send in a list with the id:
[FIELD_COMMANDS         = 0x09](https://github.com/markqvist/LXMF/blob/96dddf1b3a0328f22b59ac7d4fcc8193d9884a44/LXMF/LXMF.py#L63)

liamcottle, 2025-01-05:
Hey there! Commands aren't supported in MeshChat yet. I don't have any plans to add them in right now. Is there something specific you're trying to interact with?

Terramoto, 2025-01-05:
> Hey there! Commands aren't supported in MeshChat yet. I don't have any plans to add them in right now. Is there something specific you're trying to interact with?

Hey there, i was trying to run commands on one of the distribution groups script in https://github.com/SebastianObi/LXMF-Tools i'm currently running the extended one which accepts commands via normal messaging, but because i noticed lxmf, sideband and the script supported it i thought MeshChat would also have it or that the feature was something part of the network.