liamcottle/reticulum-meshchat #42: [Feature] Nomad Network Page Bookmarks

LinuxinaBit, 2025-01-06:
Similar to the functionality in NomadNet, the ability to save or bookmark pages in MeshChat would be very nice to have.

This could help make it harder for others to impersonate a node by just naming their node the same thing, because the identity hash and address of the real node are already saved in the 'bookmark'.
It's mostly a TOFU thing, but it's better than nothing.

LinuxinaBit, 2025-01-06:
This also opens the door for an `Identify when connecting` type option, allowing for true Reticulum identity based authentication for things like nomadForum.

brothercorvo, 2025-05-09:
support this

LinuxinaBit, 2025-07-24:
See https://github.com/markqvist/Reticulum/discussions/837 for a concept on how this could be integrated into a well-built trust system.

liamcottle, 2025-07-25:
Finally got around to this one.

- Added simple favourites system for NomadNetwork Nodes in 4734e6246833adf785aab83485136e977a1d8024
- Added ability to identify self to a NomadNetwork Node in 37cc6aa158b7e5511b28474ec05ad9e42e4cb030

Closing as completed 🚀

closed 2025-07-25 by liamcottle

LinuxinaBit, 2025-07-25:
Awesome!
Looks really good :)

u32luke, 2025-07-27:
Fantastic! TY @liamcottle