liamcottle/reticulum-meshchat #44: meshchat.py by itself or maybe docs explaining routes

resession, 2025-01-08:
is there any resource or docs that explain the websocket server and what the different routes/endpoints are for?

i work on a p2p web browser, https://github.com/HybridWare/hybrid-browser

i've been trying see how i can bring reticulum to the browser/electron. if you can explain the different routes and endpoints of the meshchat websocket server that meshchat.py creates, then it will make it easy for people to make sites and browser apps that can interact with reticulum.

i know you're working https://github.com/liamcottle/rns.js, which i am very interested in. but since meshchat uses the official main python implementation of reticulum, this will have all the features.

i'd love to create a protocol handler in electron, something like `reticulum://` and then we can grab reticulum data using your meshchat websocket server.

ducksandgoats, 2025-01-25:
what i had in mind is having a protocol scheme such as `reticulum://` that can do CRUD interactions with reticulum data on the browser. my app https://github.com/HybridWare/hybrid-browser (fork of https://github.com/AgregoreWeb/agregore-browser) is a p2p browser. it uses electron's protocol handler capabilities to bring p2p networks to the browser. for example, in a single html page, you can do `fetch('ipfs://hash/path/to/file.ext', {method: 'GET'})` and `fetch('hyper://key/some/path/to/file.ext', {method: 'POST', body: 'some text'})`.  agregore and hybrid supports multiple protocols for different networks.

this is all the protocl handler logic https://github.com/HybridWare/hybrid-browser/tree/main/app/protocols

so in my example, you can have a single html page and do `fetch('reticulum://identity/some/path/to/something')`.

i will look into RNS, maybe i can have a local server with RNS and proxy all the http requests and interact with reticulum data that way.

liamcottle, 2025-02-08:
Sounds cool! Interested to see how you get on with an implementation for that! I'll close out this issue, since there's no problem here and you can read through the source code of MeshChat if you want to see how the APIs work :)

closed 2025-02-08 by liamcottle