liamcottle/reticulum-meshchat #54: Headless mode not working with AppImage

psic4t, 2025-01-28:
./ReticulumMeshChat-v1.19.0-linux.AppImage --help  shows a "--headless" option, but it is without effect. The electron GUI is still opening.

liamcottle, 2025-02-08:
Howdy! The `--headless` option was intended to prevent launching the default web browser when running from source. This wasn't supported in the release binary with Electron.

However, I've just pushed commit 59deac6d07d4d473f380cc8c246cab308cc104a1 which should allow the next released binary to be launched with `--headless` to prevent the Electron GUI from showing.

Log output will just show in the terminal you launched it from. This has been tested on macOS so far, but not Windows or Linux.

closed 2025-02-08 by liamcottle

psic4t, 2025-02-08:
Great, thank you! :)