liamcottle/reticulum-meshchat #59: QR code of LoRa settings

joakim, 2025-01-31:
I saw something cool in a Meshtastic demo. Couldn't MeshChat also do this for LoRa parameters?

https://www.youtube.com/watch?v=IUJEs-OoeMo#t=20m31s

You can see in the demo how quick it was for people in the audience to connect.

PS. Thank you for the excellent work on making Reticulum accessible to more people! And easier to use for all of us :)

LinuxinaBit, 2025-02-03:
See https://github.com/markqvist/Reticulum/discussions/19
Probably best to wait for that.

NumbarZero, 2025-06-23:
there should really be somekind of default standard for the LoRa setting. i want to help build the Reticulum network, but everybody is using different RF settings so its impossible to be sure what to use. also its not very beginner friendly having to enter frequencies and coding rate.