liamcottle/reticulum-meshchat #61: Crash when starting after adding RNode

stephen304, 2025-02-06:
Hi, I was trying out RNode / meshchat and just finished flashing my RAK4631 using `rnodeconf --autoinstall`, I downloaded the latest appimage for meshchat and added an interface with the /dev/ttyACM0 from the RAK and set a frequency and other connection parameters. After restarting for it to take effect, I now get the error "Meshchat crashed unrecognized arguments --no-sandbox"

[image]

liamcottle, 2025-02-08:
Yeah there's some fun issues with the Linux AppImage that I haven't had a chance to look into. It does seem like you're passing in the `--no-sandbox` flag on the command line, or maybe a shortcut is adding it.

That flag is to tell Electron/Chromium to disable sandboxing, which seems to be required on some Linux machines, but for some reason the flag is passed into the MeshChat python executable as well. Probably because I pass in all arguments to it...

If you need this flag, I guess the solution is for MeshChat to just ignore invalid arguments, rather than complaining about unknown arguments...

stephen304, 2025-02-08:
I guess "integrating" the app image adds the flag, the only way to avoid it seems to be running it from my downloads folder, but that's kind of annoying because I have to click "run once" on the run dialog popup every time instead of just being able to launch it directly. Maybe known appimage flags could be stripped before passing other flags onto Python?

commit e9735271 (2025-05-11): Merge pull request #81 from stephen304/ignore-known-flatpak-flags

filter out known flags that should not be passed to python. fixes #61
commit e9a9e9f8 (2025-05-11): filter out known flags that should not be passed to python. fixes #61
closed 2025-05-11 by liamcottle