liamcottle/reticulum-meshchat #63: After a couple of hours I can't open pages any more. A reload fixes the problem.

woltere, 2025-02-07:
After having the application open for a couple of hours I can't open pages any longer.
Hitting command+R to reload the application fixes the problem.

The error in the concole I get is this:

```
[2025-02-07 18:30:02] [Error] Error while executing response received callback from <RNS.Link.RequestReceipt object at 0x11007fa10>. The contained exception was: <asyncio.locks.Lock object at 0x10e4fb950 [unlocked, waiters:1]> is bound to a different event loop
```

I don't know if it has anything to do with the screen saver being activated when I am away from the computer for a while.

liamcottle, 2025-02-07:
- If you running a release binary, which version and which OS?
- If you are running from source, which commit hash?

woltere, 2025-02-07:
Version 1.19.0
macOS 15.3 (24D60)

liamcottle, 2025-02-07:
Great, thanks!

I pushed some commits last week that should hopefully solve the threading/async issues in callbacks.
https://github.com/liamcottle/reticulum-meshchat/commit/2c518d1b31d2c9038e89d8050bae718cdabc4ae1

These changes aren't in a release binary yet, but will go out in v.1.20.0 which will hopefully be later today.

liamcottle, 2025-02-08:
[v1.20.0](https://github.com/liamcottle/reticulum-meshchat/releases/tag/v1.20.0) has been released, and should _hopefully_ resolve this. Please feel free to reopen if this happens again :)

closed 2025-02-08 by liamcottle