liamcottle/reticulum-meshchat #64: Raspberry Pi reticulum-meshchat loses interfaces when started as a dervice

fotografm, 2025-02-07:
Hi Liam, I followed your instructions to install on Raspberry Pi and it works fine. I then followed your instructions to start reticulum-meshchat as a service. Status appears normal in
sudo systemctl status reticulum-meshchat.service
After a reboot, meshchat default interface is DOWN according to rnstatus and is "disconnected" in the gui interfaces tab on a web browser. Am I missing something ? Ideally I would like everything to survive and start automatically after reboot as I plan to run it as a headless remote system.
Cheers
Simon

liamcottle, 2025-02-07:
Howdy! I would probably suggest looking at the logs for the service to see why the interface isn't coming up.

- Edit your `~/.reticulum/config` and set `loglevel = 7` for more information.
- Run `journalctl -u reticulum-meshchat -f` to watch the service logs.

fotografm, 2025-02-07:
Hi Liam, If I do "sudo service reticulum-meshchat restart" then the behaviour is as expeced and the default interface is UP. The log file looks normal. The problem is that when I reboot the raspberry pi, the reticulum-meshchat starts as expected but the default interface is DOWN. I will try looking at the logfile with level 7 after booting. Is it OK  to expect the interface to come up after a reboot ?

liamcottle, 2025-02-07:
Yes the interface should come up after a reboot. Once you have inspected the logs after a reboot, we will know more.

If you used the service file example I provided, it has the line `After=network.target`, which should wait for the network services to be ready before starting the MeshChat service.

Let me know what you find in the logs after a reboot :)

fotografm, 2025-02-08:
Hi Liam, Here is the logfile output:

Feb 08 08:59:13 raspi9 systemd[1]: Started reticulum-meshchat.service - reticulum-meshchat.
Feb 08 08:59:16 raspi9 env[572]: Reticulum Identity <886fc381a045da9143231796a0721854> has been loaded from file storage/identity.
Feb 08 08:59:16 raspi9 env[572]: Using Storage Path: storage/identities/886fc381a045da9143231796a0721854
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Debug] Started shared instance interface: Shared Instance[37428]
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Debug] Cleaning ratchets...
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Extra] Cleaning resource and packet caches...
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Verbose] Bringing up system interfaces...
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Extra] AutoInterface[Default Interface] No link-local IPv6 address configured for lo, skipping interface
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Warning] AutoInterface[Default Interface] could not autoconfigure. This interface currently provides no connectivity.
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Debug] AutoInterface[Default Interface] hardware MTU set to 1064
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Verbose] System interfaces are ready
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Verbose] Utilising cryptography backend "openssl, PyCA 44.0.0"
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Verbose] Configuration loaded from /home/fotog/.reticulum/config
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Verbose] Loaded 446 known destination from storage
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Verbose] Loaded Transport Identity from storage
Feb 08 08:59:16 raspi9 env[572]: [2025-02-08 08:59:16] [Debug] Ratchets enabled on <lxmf.delivery.886fc381a045da9143231796a0721854/0729a38637cc79fb5f26cccfe6777312>
Feb 08 08:59:16 raspi9 env[572]: ======== Running on http://0.0.0.0:8000 ========
Feb 08 08:59:16 raspi9 env[572]: (Press CTRL+C to quit)

"AutoInterface[Default Interface] No link-local IPv6 address configured for lo, skipping interface" seems to be the problem. It is a Raspberry Pi 4 running headless raspi OS 64 bit basd on Debian Bookworm connected via wifi.  ip a returns:

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:94:a2:28 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.202/24 brd 192.168.8.255 scope global dynamic noprefixroute eth0
       valid_lft 43184sec preferred_lft 43184sec
    inet6 fe80::8d4c:1e38:c590:b98f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether d8:3a:dd:94:a2:2c brd ff:ff:ff:ff:ff:ff

I tried to get a lo IPv6 address using:

sudo sysctl -w net.ipv6.conf.lo.autoconf=1
sudo ip -6 addr add fe80::1/64 dev lo

and now ip a shows:

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 fe80::1/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:94:a2:28 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.202/24 brd 192.168.8.255 scope global dynamic noprefixroute eth0
       valid_lft 42711sec preferred_lft 42711sec
    inet6 fe80::8d4c:1e38:c590:b98f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether d8:3a:dd:94:a2:2c brd ff:ff:ff:ff:ff:ff

To make the IPv6 address  survive a reboot I used systemd:
sudo nano /etc/systemd/network/10-lo-link6.network   with content:

[Match]
Name=lo

[Network]
LinkLocalAddressing=yes

[Address]
Address=fe80::1/64

After reboot, ip -6 addr show lo shows:

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    inet6 fe80::1/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever

and Meshchat GUI in webrowser shows "default interface connected" so problem is solved thanks to your help ! I included all of the above solution in case anyone else has the same problem. Please let me know if mysolution is OK or whether it is full of glaring problems and vulns.

Thanks again
Cheers
Simon

fotografm, 2025-02-08:
The GUI appears to still be connected but logfile shows:

Feb 08 09:37:08 raspi9 systemd[1]: Started reticulum-meshchat.service - reticulum-meshchat.
Feb 08 09:37:11 raspi9 env[580]: Reticulum Identity <886fc381a045da9143231796a0721854> has been loaded from file storage/identity.
Feb 08 09:37:11 raspi9 env[580]: Using Storage Path: storage/identities/886fc381a045da9143231796a0721854
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Debug] Started shared instance interface: Shared Instance[37428]
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Debug] Cleaning ratchets...
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Extra] Cleaning resource and packet caches...
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Verbose] Bringing up system interfaces...
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Extra] AutoInterface[Default Interface] Selecting link-local address fe80::1 for interface lo
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Extra] AutoInterface[Default Interface] Creating multicast discovery listener on lo with address ff12:0:d70b:fb1c:16e4:5e39:485e:31e1
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable
Feb 08 09:37:11 raspi9 env[580]: [2025-02-08 09:37:11] [Verbose] AutoInterface[Default Interface] discovering peers for 1.5 seconds...
Feb 08 09:37:12 raspi9 env[580]: [2025-02-08 09:37:12] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Debug] AutoInterface[Default Interface] hardware MTU set to 1064
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Verbose] System interfaces are ready
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Verbose] Utilising cryptography backend "openssl, PyCA 44.0.0"
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Verbose] Configuration loaded from /home/fotog/.reticulum/config
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Verbose] Loaded 446 known destination from storage
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Verbose] Loaded Transport Identity from storage
Feb 08 09:37:13 raspi9 env[580]: [2025-02-08 09:37:13] [Debug] Ratchets enabled on <lxmf.delivery.886fc381a045da9143231796a0721854/0729a38637cc79fb5f26cccfe6777312>
Feb 08 09:37:13 raspi9 env[580]: ======== Running on http://0.0.0.0:8000 ========
Feb 08 09:37:13 raspi9 env[580]: (Press CTRL+C to quit)
Feb 08 09:37:27 raspi9 env[580]: [2025-02-08 09:37:27] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable
Feb 08 09:37:28 raspi9 env[580]: [2025-02-08 09:37:28] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable
Feb 08 09:37:29 raspi9 env[580]: [2025-02-08 09:37:29] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable
Feb 08 09:37:31 raspi9 env[580]: [2025-02-08 09:37:31] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable
Feb 08 09:37:32 raspi9 env[580]: [2025-02-08 09:37:32] [Warning] AutoInterface[Default Interface] Detected possible carrier loss on lo: [Errno 101] Network is unreachable

liamcottle, 2025-02-08:
I've not ever passed a `device` to the `AutoInterface`, but my first thoughts are that you'd want to bind to something like `eth0` or `wlan0` rather than `lo` for the loop back local only interface. I might be missing something here, but that's just off the top of my head.

I generally avoid using `AutoInterface` myself, and prefer setting up `TCPServerInterface` and `TCPClientInterface` as they're more reliable. Although `AutoInterface` is nice as you don't usually have to configure anything at all.

Have you tried using `eth0` or `wlan0`, instead of `lo`? and if you do intended to use `lo`, what is the use case here?

fotografm, 2025-02-08:
Actually, the raspberry pi isnt using wifi, it's using eth0. I get confused between different machines...Next I should try binding to eth0 instead of lo. I don't have a particular use case for lo, it's just that I saw that the lack of IPv6 address for lo was causing the problem so I tried to fix it. In fact, I think I will save time by following your advice and use the server and client interfaces as you suggest instead. Please could you give such an example config for these interfaces on the raspberry pi which is intended to have an RNode connected aswell, to allow meschat machines on my network to connect via it to Lora. I would need a suitable [Interfaces] config for the pi and for the other machine connected in the same LAN. Rnode config I have got to work successfully already.

closed 2025-02-16 by fotografm

adingbatponder, 2025-02-17:
see https://www.loramesh.org/subpages/pi_install.html