liamcottle/reticulum-meshchat #72: Adroid <ip>:8000 to meshchat GUI

adingbatponder, 2025-03-05:
What work-around/s could be done to make <ip>:8000 to meshchat GUI work for anyone using a standard Android and standard device? Or special Android settings? 

There was an issue created already that was closed on the grounds that this worked for a certain device, but this is a real show-stopper for me and my device. I have now had this reported from others too. It might help to keep an issue open, or re-open the existing one, until a real work-around is public. This might help to make people realise that they are not the only ones who have the Android <ip>:8000 problem and thereby save frustration. 

I realise this is entirely a google issue, not a meshchat "bug" at all, since <ip>:8000 works fine on linux and iOS. Nevertheless, is there a way meshchat could be tweaked to force <ip>:8000, or similar, to work for all standard Android devices? 

Thanks a lot.

adingbatponder, 2025-03-07:
The problem is also described here https://youtu.be/T1itQcdf5cc?feature=shared&t=328

HotMicSystems, 2025-03-07:
Can you configure meshchat to run to the assigned local IP(ex 192.168.1.xxx)?
Example:
```
# python3 
python3 meshchat --host 192.168.1.xxx --port 8000
```
or alternatively if the issue persist, bind to all interfaces 
```
python3 meshchat --host 0.0.0.0 --port 8000
```