liamcottle/reticulum-meshchat #77: [crash] ReticulumMeshChat: error: unrecognized arguments: --no-sandbox

mix127, 2025-03-31:
I'm just starting.

When I run program on Linux Ubuntu i get:

```
./ReticulumMeshChat-v1.21.0-linux.AppImage
[36656:0331/214250.979285:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_ReticuOqbzsw/chrome-sandbox is owned by root and has mode 4755.
```

When run without sandbox (and   --disable-setuid-sandbox too) I get

```
./ReticulumMeshChat-v1.21.0-linux.AppImage  --no-sandbox
usage: ReticulumMeshChat [-h] [--host [HOST]] [--port [PORT]] [--headless]
                         [--identity-file IDENTITY_FILE]
                         [--identity-base64 IDENTITY_BASE64]
                         [--generate-identity-file GENERATE_IDENTITY_FILE]
                         [--generate-identity-base64]
                         [--reticulum-config-dir RETICULUM_CONFIG_DIR]
                         [--storage-dir STORAGE_DIR]
                         [--test-exception-message TEST_EXCEPTION_MESSAGE]
                         ...
ReticulumMeshChat: error: unrecognized arguments: --no-sandbox

[37017:0331/214524.434461:ERROR:browser_main_loop.cc(280)] GLib-GObject: ../../../gobject/gsignal.c:2684: instance '0x242000db0a50' has no handler with id '5635'

```

stephen304, 2025-04-17:
I had the same issue and noticed it works when I click "Run once" instead of "Integrate and run". So for now I just launch it from my downloads folder.

stephen304, 2025-04-18:
So it looks like this specific case is caused by 2 separate issues:

Running `./ReticulumMeshChat-v1.21.0-linux.AppImage` you seem to be having an issue related to Ubuntu's appimage restrictions. [Another dev solved it by](https://github.com/zoidsh/meru/issues/419#issuecomment-2775318588) updating the version of `electron-builder` to v26 [like so](https://github.com/zoidsh/meru/commit/ba3f1da08089a9a62c03fb5c502b35712be9e8b9)

The issue with using the `--no-sandbox` flag I think I was able to fix in my PR https://github.com/liamcottle/reticulum-meshchat/pull/81 , so you could try building the appimage from that branch using `npm install; pip install -r requirements.txt; npm run dist`

thiswillbeyourgithub, 2025-05-12:
Had the issue today too, but only when I was running rnsd. I initially thought that if i was running rnsd then meshchat would be able to use its already setup network, but no it crashed, telling me that the ports where already taken etc.

liamcottle, 2025-05-15:
`--no-sandbox` should now be working in v1.22.0+

closed 2025-05-15 by liamcottle