liamcottle/reticulum-meshchat #88: Program will not start anymore

KaiNordic, 2025-06-25:
I followed Andy´s YouTube guidance on how to flash Meshtastic on a Reticulum ESP32 V3 and added it to the Reticulum MeshChat Software which looked fine until I restarted the program:

MeshChat Crashed!

Exit Code: 255

----- stdout -----

Reticulum Identity <ce5be40662ff95fbb5a8935b3e2c8a13> has been loaded from file /Users/polar/.reticulum-meshchat/identity.
Using Storage Path: /Users/polar/.reticulum-meshchat/identities/ce5be40662ff95fbb5a8935b3e2c8a13
[2025-06-25 20:11:23] [Error]    Invalid TX power configured for RNodeInterface[SonneundWind]
[2025-06-25 20:11:23] [Error]    The interface "SonneundWind" could not be created. Check your configuration file for errors!
[2025-06-25 20:11:23] [Error]    The contained exception was: The configuration for RNodeInterface[SonneundWind] contains errors, interface is offline

----- stderr -----

TX power was set to 30 so not sure why that would caused problems. I also did not swich off or unplugged the device, so I am a bit lost here.

HELP please

KaiNordic, 2025-06-25:
Oh and one more note: I uninstalled the software and reinstalled that did not do anything same error even with the note not connected.

liamcottle, 2025-07-18:
A TX power of 30 is more than likely not supported by your radio. More likely 22dBm will be max, if not lower.

You'll need to manually edit your config file to fix this, and then MeshChat should be able to launch. You config file should be located here:

```
/Users/polar/.reticulum/config
```

Either remove the entry for `RNodeInterface[SonneundWind]`, disable it, or fix the tx power :)