liamcottle/reticulum-meshchat #92: Voice and image attachments no longer work

cybertron49, 2025-07-17:
Meshchat was working fine but after upgrading to 2.0, I can no longer send voice or image attachments to myself. I tried on both Windows and Linux. Text messages work fine. Sideband can do everything but Meshchat no longer works like before.

LinuxinaBit, 2025-07-18:
If you are trying to send them to clients using RNS versions 0.9.5 or older, that is expected.
Try a direct connection between two peers running the newest RNS versions available. If that doesn't fix it, there is likely an issue.

[image]

LinuxinaBit, 2025-07-18:
If that clears up your concerns, please consider closing this issue to help clear up Liam's 'inbox' ;)

LinuxinaBit, 2025-07-19:
What version of MeshChat are you using on both ends?