liamcottle/reticulum-meshchat #95: [Bug] Compatible NomadNet pages still fail to load sometimes

LinuxinaBit, 2025-07-23:
Every few times I launch MeshChat, I'll be browsing through NomadNet pages and nothing, not even known compatible pages will load.
It starts throwing errors like this:
`RuntimeError: <asyncio.locks.Lock object at 0x7f3e2b039d00 [unlocked, waiters:1]> is bound to a different event loop`

LinuxinaBit, 2025-07-23:
I'm not sure if this is my fault or not actually, will have to test.

LinuxinaBit, 2025-07-24:
It seems this is related to starting NomadNet while MeshChat is already running, opening a page in NomadNet, then closing it.
Not sure if this is exclusively a MeshChat related issue, but I don't think it's happened to me with NomadNet.
Note that I am using a portable installation of MeshChat.

Feel free to close this issue if it isn't relevant.

LinuxinaBit, 2025-07-25:
After more testing, I just had it happen without using any other Reticulum programs on my system...

liamcottle, 2025-07-27:
Thanks for the report. I've seen this one a few times.

Python threading/async stuff is really painful. You'll find that if you force reload the page (which opens a new websocket connection from browser to meshchat.py), the NomadNetwork pages will start loading properly again.

It seems to be related to using the wrong event loop when sending the response back to the websocket connection that requested the page.

Will need to investigate this more...

liamcottle, 2025-07-28:
I've reworked the async event handling to always use the main event loop in 65df111b876dd06baa1e4887df9e870586db3808.
This _should_ resolve the issue. Please let me know if you still run into this again after updating :)

closed 2025-07-28 by liamcottle