liamcottle/reticulum-meshchat #96: Meshchat fails to launch after unexpected power loss

DanBeard, 2025-07-25:
I've run into a situation a few times on a raspi zero 2W where after unexpected powerloss, meshchat won't launch on reboot. I think it's related to a corrupt /rachet file since a temporary workaround is to just delete all of the .rachet files in the storage directory. For example `find {MESHCHAT_PATH}/storage -type f -name '*.rachet' -delete`. After deleting the .rachet files, mesh-chat appears to launch and work normally. 

Exception traceback: 
```
[2025-07-25 21:07:59] [Error]    First ratchet reload attempt for <lxmf.delivery.4151bf80664bb48c9bcf67f69f681072:17bdc7979e4e544aa77ddc2b9aa55352> failed. Possible I/O conflict. Retrying in 500ms.
[2025-07-25 21:07:59] [Error]    An unhandled <class 'RNS.vendor.umsgpack.InsufficientDataException'> exception occurred: 
[2025-07-25 21:07:59] [Error]    Traceback (most recent call last):
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/Destination.py", line 452, in _reload_ratchets
    load_attempt()
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/Destination.py", line 442, in load_attempt
    persisted_data = umsgpack.unpackb(ratchets_file.read())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/vendor/umsgpack.py", line 1131, in _unpackb3
    return _unpack(io.BytesIO(s), options)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/vendor/umsgpack.py", line 936, in _unpack
    code = _read_except(fp, 1)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/vendor/umsgpack.py", line 707, in _read_except
    raise InsufficientDataException()
RNS.vendor.umsgpack.InsufficientDataException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/Destination.py", line 458, in _reload_ratchets
    load_attempt()
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/Destination.py", line 442, in load_attempt
    persisted_data = umsgpack.unpackb(ratchets_file.read())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/vendor/umsgpack.py", line 1131, in _unpackb3
    return _unpack(io.BytesIO(s), options)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/vendor/umsgpack.py", line 936, in _unpack
    code = _read_except(fp, 1)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/retcon/retcon/venv/lib/python3.11/site-packages/RNS/vendor/umsgpack.py", line 707, in _read_except
    raise InsufficientDataException()
RNS.vendor.umsgpack.InsufficientDataException
```

liamcottle, 2025-07-27:
Hey, thanks for the report. This crash happens inside the RNS library itself, and would need to be patched there. Unfortunately MeshChat can't do anything if the RNS stack fails to start up.

DanBeard, 2025-07-27:
Does this happen before meshchat is even loaded? Does meshchat send the rachets files to RNS somehow? I'm trying to figure out if there maybe some sanity check to the rachets files meshchat could do before passing them to rns? Since deleting the files in meshchat's storage folder fixes the issue in RNS. 

I keep running into it since people tend to just pull the power on the raspi zeros I'm handing out with meshchat on them 😅 Adding `find {MESHCHAT_PATH}/storage -type f -name '*.ratchets' -delete` to every startup seems to be an ok workaround for now, but I'd like to fix it more elegantly somehow.

adingbatponder, 2025-07-28:
Interestingly I have a pi zero 2W running for months and restart it for all sorts of reasons and it works flawlessly on restart... I will keep an eye on this as re-start after unexpected power loss is major requirement for my solar based projects. Thanks for this work-around, which would seem good enough for my purposes.